prusa3d/PrusaControl #3: Advanced mode to change Slic3r settings?

ahalekelly, 2017-06-08:
I like the GUI, but have tweaked my Slic3r settings a lot to get them where I like them. I wouldn't use any software unless it offered me the same configuration options. Even an option to use one of the Slic3r settings files without any way of changing them in Prusa Control, like Repetier Host does, would be usable.

daktylus, 2017-08-14:
I am pretty sure (and I would love that) that its their plan to develop prusa control to a comprehensive fronted for SLicer Prusa edition.

boelle, 2017-09-19:
i tried to change the json file where the settings are stored

but it then failed to start

the settings are here: C:\Program Files (x86)\PrusaControl\data