prusa3d/PrusaControl #9: PrusaControl doesn't start

coolcow, 2017-06-11:
On a fresh installation of Windows 10 (64bit) in a VM i get this error when i start PrusaControl.

[image]

The error `Failed to execute script main` stays in the backgroud behind the splashscreen.
I have uninstalled and installed PrusaControl again a couple of times, but without success.

Slic3r PE works fine.

What am i doing wrong ?

tibor-vavra, 2017-06-12:
Hi,
attach a log file from PrusaControl. It is placed in C:\Users\{your user name}\prusacontrol.log

coolcow, 2017-06-12:
It said:
```
[INFO][controller.py:55:__init__()]-Local path: C:\Program Files (x86)\PrusaControl\
[ERROR][main.py:77:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 170, in <module>
  File "main.py", line 139, in main
  File "controller.py", line 62, in __init__
  File "parameters.py", line 283, in __init__
  File "configparser.py", line 696, in read
  File "configparser.py", line 1077, in _read
configparser.MissingSectionHeaderError: File contains no section headers.
file: 'C:\\Users\\jruiz\\prusacontrol.cfg', line: 1
'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

After deleting `prusacontrol.cfg' it works now.
Maybe the application should ask if it should create a new configuration file (in the case of a corrupted one) instead of just crashing. But hey, its Beta ! :)

Thx for the hint !

Ryans585, 2017-09-15:
I am receiving the same Error message on PrusaControl launch. My error log says

[ERROR][main.py:95:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 193, in <module>
  File "main.py", line 159, in main
  File "controller.py", line 65, in __init__
  File "parameters.py", line 354, in __init__
  File "parameters.py", line 477, in check_versions
  File "parameters.py", line 514, in get_materials_info
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Ryan\\.prusacontrol\\materials_i3_mk2mm.json'

wootner, 2017-09-15:
Has been working fine ...tried removing `prusacontrol.cfg' didn't work
Uninstalled and reinstalled after reboot worked 

[INFO][controller.py:55:__init__()]-Local path: C:\Program Files (x86)\PrusaControl\
[ERROR][main.py:77:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 170, in <module>
  File "main.py", line 139, in main
  File "controller.py", line 62, in __init__
  File "parameters.py", line 293, in __init__
  File "parameters.py", line 407, in check_versions
  File "parameters.py", line 446, in get_materials_info
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\jason\\.prusacontrol\\materials_i3_mk2mm.json'

Ryans585, 2017-09-15:
I copied the 'materials_i3_mk2.json' file and renamed it 'materials_i3_mk2mm.json'. it launches just fine now.

wootner, 2017-09-15:
Yep that works

Many thanks

ljcousse, 2018-09-24:
I have the same issue. Logfile gives this output:

[ERROR][main.py:95:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 193, in <module>
  File "main.py", line 159, in main
  File "controller.py", line 65, in __init__
  File "parameters.py", line 357, in __init__
  File "parameters.py", line 497, in check_versions
  File "parameters.py", line 576, in get_materials_info
  File "json\__init__.py", line 265, in load
  File "c:\python35\lib\encodings\cp1252.py", line 23, in decode
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 54870: character maps to <undefined>

RogerStr, 2018-09-24:
> 
> 
> I have the same issue. Logfile gives this output:
> 
> [ERROR][main.py:95:log_exception()]-Logging an uncaught exception
> Traceback (most recent call last):
> File "main.py", line 193, in
> File "main.py", line 159, in main
> File "controller.py", line 65, in **init**
> File "parameters.py", line 357, in **init**
> File "parameters.py", line 497, in check_versions
> File "parameters.py", line 576, in get_materials_info
> File "json__init__.py", line 265, in load
> File "c:\python35\lib\encodings\cp1252.py", line 23, in decode
> UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 54870: character maps to
I have the exact same problem

thiaggo, 2018-09-24:
same error here  i think it has to be with the new sla printer that came out

[ERROR][main.py:95:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 193, in <module>
  File "main.py", line 159, in main
  File "controller.py", line 65, in __init__
  File "parameters.py", line 357, in __init__
  File "parameters.py", line 497, in check_versions
  File "parameters.py", line 576, in get_materials_info
  File "json\__init__.py", line 265, in load
  File "c:\python35\lib\encodings\cp1252.py", line 23, in decode
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 54870: character maps to <undefined>

Voodazel, 2018-09-24:
Same issue

[ERROR][main.py:95:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 193, in <module>
  File "main.py", line 159, in main
  File "controller.py", line 65, in __init__
  File "parameters.py", line 357, in __init__
  File "parameters.py", line 497, in check_versions
  File "parameters.py", line 576, in get_materials_info
  File "json\__init__.py", line 265, in load
  File "c:\python35\lib\encodings\cp1252.py", line 23, in decode
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 54870: character maps to <undefined>

moeskerv, 2018-09-24:
Same here
[ERROR][main.py:95:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 193, in <module>
  File "main.py", line 159, in main
  File "controller.py", line 65, in __init__
  File "parameters.py", line 357, in __init__
  File "parameters.py", line 497, in check_versions
  File "parameters.py", line 576, in get_materials_info
  File "json\__init__.py", line 265, in load
  File "c:\python35\lib\encodings\cp1252.py", line 23, in decode
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 54870: character maps to <undefined>

probonopd, 2018-09-24:
To everyone who has the same issue: Please report your exact operating system and version, thanks.

moeskerv, 2018-09-24:
Win10 PrusaControl 0.9.4_415 beta - but issue seems to be fixed now.

moeskerv, 2018-09-24:
It seems that https://github.com/prusa3d/PrusaControl-settings/commit/e53b853e22f6bedbc4bf26d68d8193ea8933233f fixed the issue.

schnaggels, 2018-11-03:
Had the same error also with this version...
The reason was simple. 

I did hide these 2 files as I didn't want to see them in my main user folder.
C:\Users\{your user name}\prusacontrol.log
C:\Users\{your user name}\prusacontrol.cfg

Usually works for other apps but not here, after removing windows hide file flag the startup worked as before...

closed 2026-02-26 by coolcow