prusa3d/PrusaControl #14: Provide an AppImage for Linux

probonopd, 2017-06-12:
Providing an [AppImage](http://appimage.org/) (like the one that is provided for [Slic3r Prusa Edition](https://github.com/prusa3d/Slic3r/releases)) would have, among others, these advantages:
- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions

__Would there be interest in this?__

tibor-vavra, 2017-06-13:
I like this idea.

At this moment I have maximal priority on creating build for OsX, but I'll be glad for any help with linux build

probonopd, 2017-06-13:
@tibor-vavra how can you best be reached, are you on IRC?

tibor-vavra, 2017-06-14:
No, Im not. You can send me email on tibor (a) prusa3d.cz or use hangout

probonopd, 2017-07-01:
Thank you very much for providing an AppImage for download on http://prusacontrol.org/

closed 2017-07-01 by probonopd