prusa3d/PrusaControl #15: Feature request: Send to Octoprint

yschroeder, 2017-06-17:
I don't know if you want PrusaControl be a full-blown controll center to directly communicate with the printer (the same suggests this). However, I'd lke to send my G-code directly to Octoprint, as it is possible with Slic3r.

quibusus, 2017-07-05:
+1, this would be an awesome feature

cwestpha, 2017-08-01:
A million times +1 ^.~

probonopd, 2017-08-28:
:+1: In fact I have started implementing this some time ago in https://github.com/prusa3d/PrusaControl/compare/ca7cedf...probonopd:WirelessPrinting - and am using it on a regular basis ever since.

An AppImage for testing is available at
https://bintray.com/probono/AppImages/PrusaControl_Wireless/1556.glibc2.17#files

This so far successfully discovers one or more printers on the network, and lets the user print to them.

[image]

Please note: The current implementation does not work with genuine OctoPrint hosts yet because the API key is currently not sent (I am using it with my own firmware that implements a custom subset of the OctoPrint protocol without API keys). OctoPrint API key sending should be relatively easy to add though, but I didn't know where to put this in the GUI. In fact, I think the whole way we enter per-printer settings should be made dynamic, based on the printers discovered on the network. Input welcome.

__Feedback welcome from @tibor-vavra and anyone interested!__ Also, Pull requests are welcome against my branch https://github.com/probonopd/PrusaControl/tree/WirelessPrinting

yschroeder, 2017-09-26:
This get more urgent, now the new MK3 has Octoprint integration.

probonopd, 2018-01-06:
Ping - are there any plans to merge this?

cnf, 2018-04-29:
I would very much like this. i like the fast and easy use of prusacontrol, but then having to manually juggle gcode files feels slow after cura "print on octoprint"

kopper42, 2019-01-19:
Absolutely a +1.  This is what keeps me from using PrusaControl.

probonopd, 2019-03-08:
https://github.com/prusa3d/PrusaControl/commit/7c9877c93e5fe319de1c0635ffbe0188b0575fe6 does implement this, but unfortunately there is no continuous build of it available to test.