prusa3d/PrusaControl #19: Potential issue with infil calculation

mikebourgeault, 2017-06-28:
I’m not 100% sure, as I didn’t test with slic3r prusa edition to compare (yet), but I ran into some issues printing from an stl file at draft quality. At first it was going well, but once my print got to where it needed to do infill (the infill portion was very small, probably just at the boundary of actually needing infill) the print got really messy. Almost as if there was way to much material being extruded for the infill (I think I was at 5% infill but don't remember, sorry). Tried 2 different filament brands and got same issue. If it helps this is what I was printing: https://www.thingiverse.com/thing:1214311

The issue started right after it finished the base, around the 3.35 layer in 0.35 mode. 

I ran out of time yesterday, but once I have some I will try it with Slic3r to see if I get same results.

prusa3d-bb, 2017-06-28:
Hello,

could you send us a picture of the issue so that we can determine whats causing it.

Best regards

Douglas

—

--- original message ---

--- end of original message ---

mikebourgeault, 2017-06-28:
Hi Douglas. I only have one picture on my phone of it, so I'll send that. I'll take a better one when I get home from work too that should show it better. I have a half completed print in white that shows the cross section. Also I plan to try the print with gcode generated from slic3r pe just to confirm it isn't something weird in the design of the downloaded item. 

Mike Bourgeault

prusa3d-bb, 2017-06-28:
Ok so we will wait to see pictures and then get back to you.

Regads

Douglas

—

--- original message ---

—

--- end of original message ---

mikebourgeault, 2017-06-29:
Hi Douglas,

I had some time to retry printing the same stl from slic3r and got the
exact same results. It looks like its an issue in the design of the object,
rather than the infill generation in the prusa control software.

bubnikv, 2017-06-30:
Prusa Control calls Slic3r for the print path generation behind the scenes.
If you experience an issue with the G-code, you should better create an
issue at the Slic3r issue tracker:

https://github.com/prusa3d/Slic3r/issues

Also please note that the issue may have already been reported, so it helps
us, if you try to find a similar issue in the tracker first and attach your
findings and data to the existing issue before entering a new issue. Thank
you.