prusa3d/PrusaControl #21: Display grams in addition to meters

probonopd, 2017-07-01:
In the "print info" box, please display grams (g) in addition to meters. This makes it easier to estimate the cost of the material used.

prusa3d-bb, 2017-07-03:
Hello,

thank you for the e-mail. I will pass this information to our developers :-)

Kind regards,

Max Kaulfuss
Junior Print Tester

PRUSA Research
+420 222 263 718 
+421 220 570 305
Partyzánská 188/7a, 170 00 , Prague
shop.prusa3d.com

--- original message ---

--- end of original message ---

mrjohnc, 2017-10-19:
This will have to be calculated for each material, here is a guide, although there are no references

https://imaterialise.helpjuice.com/materials/density

prusa3d-bb, 2017-10-24:
Hi John,

We already implemented a cost calculation tool in our Slic3r under "filament settings > filament", just under the filament diameter you can set density and cost!

Regards

Michele Moramarco
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 7-Holešovice

shop.prusa3D.com

--- original message ---

https://imaterialise.helpjuice.com/materials/density

cjander, 2018-03-27:
I really would like to have this feature build in PrusaControl as well. even for the Prusa filaments sold in your store, you are selling them per kg. there is no information about how many meters are on one spool, so how should we as users know if you don't even give us this information for your own products?

rkarlsba, 2020-09-14:
It would be rather nice to have the numbers in grams instead of meters. I rarely think of the length of filament and it'd be a PITA to measure out that before printing. Could you add this, please?

MasterCATZ, 2021-01-20:
I swear it used to show grams now all I see is metres and mm3

it is a lot easier to weigh a spool and guess remaining amount then keeping track of metres ...