prusa3d/PrusaControl #22: Does not check if object is within printable area when modifying object settings

NikWillOrStuff, 2017-07-02:
version 0.9.2_297_beta, on windows

if you import an object, then move/scale it outside of the printable area using the "Object settings" menu, then avoid moving your mouse inside the 3D preview area when you click the "Generate" button, it will start generating gcode while simultaneously giving you the "Object blah... is out of printable area!" error.

compared to the normal behavior, moving the object outside of the print area by dragging the object in the 3D view, it will instantly give you the error message, and make the "Generate" button unclickable.

this isn't an important issue, as the error message is still there in the gcode preview menu, and it makes the "Save G-Code" button unclickable.

prusa3d-bb, 2017-07-04:
Hello, 

thank you for feedback I will pass it to Prusacontrol developers, I tried to replicate it and I understand what you mean, but it does not really seem to be an important issue as it is doing what it should - preventing user from printing something out of bounds. 

Best regards,

Honza Vysušil
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
Partyzánská 188/7a, 17000, Prague
shop.prusa3D.com

--- original message ---

--- end of original message ---

tibor-vavra, 2017-07-10:
Solved by commit 4e18eb4bc739cac5c176202eb24a2487cd519acb

closed 2017-07-10 by tibor-vavra