prusa3d/PrusaControl #23: Crashes on Kubuntu 17.04

quibusus, 2017-07-05:
Starts OK, but crashes after selecting anything from the menu

```
Cannot mix incompatible Qt library (version 0x40807) with this library (version 0x40806)
Aborted (core dumped)
```

GovanifY, 2017-08-14:
dev-python/PyQt4-4.12.1 is installed on my system fwiw

GovanifY, 2017-08-14:
[image]
Yup indeed an AppImage issue, latest libs on my Gentoo runs fine

probonopd, 2017-08-14:
> Starts OK, but crashes after selecting anything from the menu
> 
> Cannot mix incompatible Qt library (version 0x40807) with this library (version 0x40806)
Aborted (core dumped)

@quibusus does this also happen with a Kubuntu 17.04 Live ISO?

Something is trying to load Qt from the system, which will not work. @tibor-vavra please let me know if I should investigate this.

tibor-vavra, 2017-08-15:
@probonopd Yes please, I think you know more about AppImage and linux distribution in general than I

probonopd, 2017-08-28:
Let's debug this:

```
me@host:~$ LD_DEBUG=libs /home/me/Downloads/prusacontrol_0.9.2.297_beta.AppImage 2>&1 | grep /usr/lib/x86_64-linux-gnu/libQ
      4750:       trying file=/tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtCore.so.4
      4750:     calling init: /tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtCore.so.4
      4750:       trying file=/tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtGui.so.4
      4750:     calling init: /tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtGui.so.4
      4750:       trying file=/tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4
      4750:     calling init: /tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4
      4750:       trying file=/tmp/.mount_prusacj0etEd/usr/lib/x86_64-linux-gnu/libQtSvg.so.4
      4750:       trying file=/usr/lib/x86_64-linux-gnu/libQtSvg.so.4
      4750:     calling init: /usr/lib/x86_64-linux-gnu/libQtSvg.so.4
```

As we can see, `libQtSvg.so.4` (wrongly) gets loaded from the system, whereas the rest of Qt (correctly) gets loaded from the AppImage (auto-mounted at `/tmp/.mount_prusacj0etEd` in this example - will be different for you).

Now, let's see whether `libQtSvg.so.4` is part of the AppImage:

```
me@host:~$ /home/me/Downloads/prusacontrol_0.9.2.297_beta.AppImage --appimage-extract                                                   
squashfs-root/.DirIcon
squashfs-root/AppRun
(...)
squashfs-root/PrusaControl-master/tools/Slic3r-Lite/bin/lib/site_perl
squashfs-root/PrusaControl-master/tools/Slic3r-Lite/bin/lib/site_perl/5.22.0
squashfs-root/PrusaControl-master/tools/Slic3r-Lite/bin/lib/site_perl/5.22.0/B.pm
fopen error
```

Oops, this looks like an (unrelated) AppImage bug. Need to investigate. In the meantime, let's use

```
me@host:~$ /home/me/Downloads/prusacontrol_0.9.2.297_beta.AppImage --appimage-mount &
/tmp/.mount_prusacgzGkz5
me@host:~$ find /tmp/.mount_prusacgzGkz5 | grep libQtSvg
(nothing)
```

So `libQtSvg.so.4` is missing from the AppImage whereas it should be bundled.

Next up, I will try to produce a fixed AppImage.

probonopd, 2017-08-28:
Please try
https://bintray.com/probono/AppImages/download_file?file_path=PrusaControl-1555.glibc2.17-x86_64.AppImage

Recipe used:
https://github.com/AppImage/AppImages/commit/a9b15d600f33f7b97811116eae6479225f2dd081

tibor-vavra, 2017-09-25:
I can confirm that new build for linux 0.9.3_390_beta(from prusacontrol.org) is function on Kubuntu 17.04. As @probonopd find, problem was missing libQtSvg library in AppImage package.

closed 2017-09-25 by tibor-vavra