prusa3d/PrusaControl #28: Extruder moves quickly over small objects, removing them from bed

probonopd, 2017-07-17:
Extruder moves quickly over small objects, removing them from the bed.

It should go around the printed objects instead.

[image]

[example.tar.gz](https://github.com/prusa3d/PrusaControl/files/1153680/example.tar.gz)

probonopd, 2017-07-17:
In Slic3r, this seems to be
Print Settings -> Layers and perimeter -> Avoid crossing perimeters

bubnikv, 2017-07-18:
Also we have the Z-hop enabled by default, so the extruder is supposed to
skip over these warping objects.

The "Avoid crossing perimeters" feature in Slic3r is not very good as of
now, it often extends the travel paths too much, leading not only to
extended print times, but it may also have adverse effects on print quality
due to extruder leakage for some materials.

jindrichbenes, 2017-07-19:
As bubnikv said, z-hop is on to avoid collisions on objects.

The Brim settings would be good to turn on for this print.

We are working on improving the auto analysis for cases such as this one.