prusa3d/PrusaControl #30: pip install PyOpenGL-accelerate on Win 10 64-bit fails

3d-gussner, 2017-07-24:
I have an issue installing PyOpenGL-accelerate on my Win 10 64-bit, does anyone has any instructions how to install all requirements on Win 10 64-bit?

3d-gussner, 2017-07-24:
Solved following these some of these instructions:
https://stackoverflow.com/questions/44437357/fatal-error-c1083-cannot-open-include-file-basetsd-h
https://stackoverflow.com/questions/35215971/lnk1158-cannot-run-rc-exe-x64-visual-studio/

- Downloaded and installed the Win 10 SDK https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk/
- Copied 'rc.exe' and 'rcdll.dll' from C:\Program Files (x86)\Windows Kits\10\bin\10.0.15063.0\x64 to C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64

3d-gussner, 2017-07-26:
I can compile without errors but the execute doesn't work :-(

3d-gussner, 2017-08-01:
@tibor-vavra sorry bordering you, but what i am doing wrong?

tibor-vavra, 2017-08-08:
Main development is made on Windows 10 64bit version. I have no problems with instalation of PyOpenGL-accelerate. 

What version of python and pip are you using? You can work without PyOpengl-accelerate but result can be really slow.

3d-gussner, 2017-08-08:
@tibor-vavra Thanks for answering...as i wrote above i got all python things installed.

My problem is when i compile it the executable doesn't work and gives me error:
---------------------------
Fatal Error!
---------------------------
Failed to execute script main

---------------------------
OK   
---------------------------

Here the python version and pip list.
Python 3.5.3 (v3.5.3:1880cb95a742, Jan 16 2017, 16:02:32)
Cython (0.26)
future (0.16.0)
multipledispatch (0.4.9)
nine (1.0.0)
numpy (1.13.1)
numpy-stl (2.2.3)
pip (9.0.1)
PyInstaller (3.2.1)
PyOpenGL (3.1.0)
PyOpenGL-accelerate (3.1.0)
pypiwin32 (219)
PyQt5 (5.9)
pyrr (0.9.1)
pystan (2.16.0.0)
python-utils (2.1.0)
setuptools (28.8.0)
sip (4.19.3)
six (1.10.0)