prusa3d/PrusaControl #36: PrusaControl crash on bootup

Dzym, 2017-08-12:
PrusaControl fails on bootup with a Fatal Error! when using Windows 8.1

Failed to execute script main.

The same installer at another windows 8.1 laptop works fine.

`
[INFO][controller.py:55:__init__()]-Local path: C:\Program Files (x86)\PrusaControl\
[ERROR][main.py:77:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "urllib\request.py", line 1254, in do_open
  File "http\client.py", line 1107, in request
  File "http\client.py", line 1152, in _send_request
  File "http\client.py", line 1103, in endheaders
  File "http\client.py", line 934, in _send_output
  File "http\client.py", line 877, in send
  File "http\client.py", line 1253, in connect
  File "http\client.py", line 849, in connect
  File "socket.py", line 712, in create_connection
  File "socket.py", line 703, in create_connection
TimeoutError: [WinError 10060] Se produjo un error durante el intento de conexión ya que la parte conectada no respondió adecuadamente tras un periodo de tiempo, o bien se produjo un error en la conexión establecida ya que el host conectado no ha podido responder

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "main.py", line 170, in <module>
  File "main.py", line 139, in main
  File "controller.py", line 62, in __init__
  File "parameters.py", line 292, in __init__
  File "parameters.py", line 363, in download_new_settings_files
  File "urllib\request.py", line 163, in urlopen
  File "urllib\request.py", line 466, in open
  File "urllib\request.py", line 484, in _open
  File "urllib\request.py", line 444, in _call_chain
  File "urllib\request.py", line 1297, in https_open
  File "urllib\request.py", line 1256, in do_open
urllib.error.URLError: <urlopen error [WinError 10060] Se produjo un error durante el intento de conexión ya que la parte conectada no respondió adecuadamente tras un periodo de tiempo, o bien se produjo un error en la conexión establecida ya que el host conectado no ha podido responder>`

Dzym, 2017-08-14:
Looks like it is not possible to download the file:
https://raw.githubusercontent.com/prusa3d/PrusaControl-settings/master/printers.json

Even when using chrome in the failing machine.

tibor-vavra, 2017-08-14:
Thats strange, do you have some proxy in your network? At start it detect if internet connection is available and when yes its start downloading actual version of printing settings.

As a temporarily workaround you can **disable your internet connection**, run PrusaControl and disable downloading of new printing settings by **turn off switch in menu Settings->PrusaControl settings->Auto update parameters**. After this you can **enable your internet connection**.

leshuguets, 2017-09-11:
Same problem here on Windows 10
Start PrusaControl 0.92_297 .... bombs  with the Failed to execute main ... error
do same thing  4 times bombs each time 
5th attempt starts and runs OK !!!!!!!!

there are problems with the SLICER gcode as it leaves some horizontal  gaps at  some z positions
the original Prusa Slic3r program interprets the gcode correctly

fjn777, 2017-11-09:
Same here with Windows 10, to me it looks like there is something wrong with .prusacontrol folder that's created under the Users folder. Seems like a permissions error