prusa3d/PrusaControl #38: Output to gcode file stops randomly

daktylus, 2017-08-14:
I loaded a gcode file (created using slicer PE) and added three color changes with prusa control. After pressinge save gcode and selecting the file name, the new gcode file is created but the output stops at some random position in the gcode. There should be another 500 lines of gcode. The Prusa GUI status bar in the lower right stops at 60%. The last line in the code is some random movement, not the position of an expected color change.

Prusa Control Version 0.9.2_297_beta
Win 10

tibor-vavra, 2017-08-15:
Can you attach the gcode file?

daktylus, 2017-08-15:
Repeated the process with a small Benchy. Same issue. Files aare attached

[3DBenchy_gcode.txt](https://github.com/prusa3d/PrusaControl/files/1225528/3DBenchy_gcode.txt)
[3DBenchy_PrusaControl_gcode.txt](https://github.com/prusa3d/PrusaControl/files/1225527/3DBenchy_PrusaControl_gcode.txt)

tibor-vavra, 2017-08-23:
Thats strange behavior, I was tried to load 3DBenchy_gcode.txt(after renaming to 3DBenchy.gcode) to PrusaControl, insert 3 color change and save. After this changes the result file was bigger of 18 bytes, thats correct.

Can you attach prusacontrol.log after fail run of PrusaControl? File is placed in your user folder(C:\users\(your user name)\)

daktylus, 2017-08-23:
[prusacontrol_log.txt](https://github.com/prusa3d/PrusaControl/files/1246253/prusacontrol_log.txt)
[3DBenchyPruConOutput_gcode.txt](https://github.com/prusa3d/PrusaControl/files/1246258/3DBenchyPruConOutput_gcode.txt)