prusa3d/PrusaControl #39: Servers down? HTTP error 500: internal server error

jokohoko, 2017-08-23:
When trying to start prusacontrol i get an error straight away.
When looking into the log file ist says this:

`[INFO][controller.py:55:__init__()]-Local path: C:\Program Files (x86)\PrusaControl\
[ERROR][main.py:77:log_exception()]-Logging an uncaught exception
Traceback (most recent call last):
  File "main.py", line 170, in <module>
  File "main.py", line 139, in main
  File "controller.py", line 62, in __init__
  File "parameters.py", line 292, in __init__
  File "parameters.py", line 378, in download_new_settings_files
  File "urllib\request.py", line 163, in urlopen
  File "urllib\request.py", line 472, in open
  File "urllib\request.py", line 582, in http_response
  File "urllib\request.py", line 510, in error
  File "urllib\request.py", line 444, in _call_chain
  File "urllib\request.py", line 590, in http_error_default
urllib.error.HTTPError: HTTP Error 500: Internal Server Error`

I searched trough other errors here on the git and saw the [link](https://raw.githubusercontent.com/prusa3d/PrusaControl-settings/master/printers.json) others where send to check, and also this gives a 500 error.

tibor-vavra, 2017-08-24:
Yes, I replicate this error yesterday. Something was wrong in github.com and they solve it after +-30 minutes. 

But its wrong, that PrusaControl cant be started without check/update of new configuration files.

tibor-vavra, 2017-09-25:
This bug was solved in build 0.9.3_390_beta.

closed 2017-09-25 by tibor-vavra