prusa3d/PrusaControl #40: Save project don't save on Linux

marcopiraccini, 2017-09-01:
Using Ubuntu 16.04, PrusaControl 0.9.2_297_beta.
"Save Project" actually saves nothing...it just creates a 21 bytes file. If you open it with "Open Project", nothing is loaded.

antoneliasson, 2017-09-15:
Saving works in 0.9.3. Opening is broken though. Opening in 0.9.3 produces a similar backtrace to what saving did in 0.9.2. Compare:

[5.1.1]anton@balder:~/Downloads> ./prusacontrol_0.9.2.297_beta.AppImage 
/bin/bash: /tmp/.mount_prusacpcp4Dy/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
/bin/bash: /tmp/.mount_prusacpcp4Dy/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
Traceback (most recent call last):
  File "./PrusaControl-master/controller.py", line 806, in save_project_file
  File "./PrusaControl-master/controller.py", line 880, in save_project
  File "./PrusaControl-master/projectFile.py", line 40, in save
  File "./PrusaControl-master/projectFile.py", line 147, in save
  File "./usr/lib/python3.4/xml/etree/ElementTree.py", line 1126, in tostring
  File "./usr/lib/python3.4/xml/etree/ElementTree.py", line 761, in write
  File "./usr/lib/python3.4/contextlib.py", line 59, in __enter__
  File "./usr/lib/python3.4/xml/etree/ElementTree.py", line 833, in _get_writer
LookupError: unknown encoding: us-ascii

[5.1.1]anton@balder:~/Downloads> ./prusacontrol_0.9.3.388_beta.AppImage 
/bin/bash: /tmp/.mount_prusacJnsmSw/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
/bin/bash: /tmp/.mount_prusacJnsmSw/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
GUI set normal support settings
GUI set normal support settings
Traceback (most recent call last):
  File "./PrusaControl-master/controller.py", line 1115, in open_project_file
  File "./PrusaControl-master/controller.py", line 1223, in import_project
  File "./PrusaControl-master/projectFile.py", line 37, in __init__
  File "./PrusaControl-master/projectFile.py", line 84, in load
  File "./usr/lib/python3.4/zipfile.py", line 937, in __init__
  File "./usr/lib/python3.4/zipfile.py", line 1016, in _RealGetContents
LookupError: unknown encoding: cp437

marcopiraccini, 2017-09-19:
Well, now save works with 0.9.3_390_beta
But opening don't, here the error (Ubuntu 16.04):
```
/bin/bash: /tmp/.mount_prusacMJUc2D/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
/bin/bash: /tmp/.mount_prusacMJUc2D/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
GUI set normal support settings
GUI set normal support settings
Traceback (most recent call last):
  File "./PrusaControl-master/controller.py", line 1115, in open_project_file
  File "./PrusaControl-master/controller.py", line 1223, in import_project
  File "./PrusaControl-master/projectFile.py", line 37, in __init__
  File "./PrusaControl-master/projectFile.py", line 84, in load
  File "./usr/lib/python3.4/zipfile.py", line 937, in __init__
  File "./usr/lib/python3.4/zipfile.py", line 1016, in _RealGetContents
LookupError: unknown encoding: cp437
````

marcopiraccini, 2017-11-13:
Testing with brand new 0.9.4. The error is still there. here the log when trying to open a project

```
⇒ ./PrusaControl_0.9.4_415_beta_64bit.AppImage       
/bin/bash: /tmp/.mount_PrusaC2h9IF1/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
/bin/bash: /tmp/.mount_PrusaC2h9IF1/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
Dpi coef: 1.0
Dpi scale: 0
Sirka praveho panelu: 250
GUI set normal support settings
GUI set normal support settings
Traceback (most recent call last):
  File "./PrusaControl-master/controller.py", line 1158, in open_project_file
  File "./PrusaControl-master/controller.py", line 1266, in import_project
  File "./PrusaControl-master/projectFile.py", line 39, in __init__
  File "./PrusaControl-master/projectFile.py", line 86, in load
  File "./usr/lib/python3.4/zipfile.py", line 937, in __init__
  File "./usr/lib/python3.4/zipfile.py", line 1016, in _RealGetContents
LookupError: unknown encoding: cp437
```

tibor-vavra, 2017-11-28:
Hi, problem was in AppImage build script. I replaced PrusaControl_0.9.4_415_beta_64bit.AppImage in release 0.9.4 by corrected version. Please try it.

marcopiraccini, 2017-12-02:
It works! Tnx.

closed 2017-12-02 by tibor-vavra