prusa3d/PrusaControl #41: Software bug?

Tommie1972, 2017-09-04:
[skallemugg 2.stl.zip](https://github.com/prusa3d/PrusaControl/files/1274833/skallemugg.2.stl.zip)

[skallemugg_2_PRUSA_PLA_DRAFT.gcode.zip](https://github.com/prusa3d/PrusaControl/files/1274816/skallemugg_2_PRUSA_PLA_DRAFT.gcode.zip)
[image]
Something went wrong in prusa control i guess when my print turns out like this?

antoneliasson, 2017-09-15:
Yes, it's a junk STL. In other words, a non-manifold STL. Slic3r will tell you this (bottom right corner of the Preview screen after slicing). Fix it in the source file or go at it with MeshMixer or similar program.

I think it is a bug in PrusaControl that it does not tell the user this. It's a basic feature I expect from the Analysis feature. In the logs I see many errors similar to

['Unable', 'to', 'close', 'this', 'loop', 'having', '266', 'points']
['Unable', 'to', 'close', 'this', 'loop', 'having', '4', 'points']
['Unable', 'to', 'close', 'this', 'loop', 'having', '1', 'points']
['Unable', 'to', 'close', 'this', 'loop', 'having', '254', 'points']
['Unable', 'to', 'close', 'this', 'loop', 'having', '4', 'points']

I recommend that the title of this bug is changed to "Analysis should warn if an object is non-manifold"

tibor-vavra, 2017-09-25:
Definitely its not a software bug, but I agree with need of new analysis for consistent of model.

Tommie1972, 2017-09-25:
Funny thing is that it slices just fine in Cura and i was able to print it that way.