prusa3d/PrusaControl #45: Failed to execute Scrpit Main / PrusaControl 0.9.3_384_beta

Stevuh, 2017-09-15:
I installed the new PrusaControl 0.9.3_384_beta and then i got the Error "Failed to execute Main".
It worked with the Version before this.
I Attach the Log file.

[prusacontrol.zip](https://github.com/prusa3d/PrusaControl/files/1306157/prusacontrol.zip)

tibor-vavra, 2017-09-15:
Hi,

Yes, I found this problem too.

As a workaround you can do this:

Download new configuration files from github.com/prusa3d/
prusacontrol-settings and place them to .prusacontrol folder.

Stevuh, 2017-09-15:
Hi,

I Downloaded and placed it in .prusacontrol folder.
But that didnt worked.

Log file below.
[prusacontrol.zip](https://github.com/prusa3d/PrusaControl/files/1306191/prusacontrol.zip)

marcopiraccini, 2017-09-15:
I have the same problem (Ubuntu 16.04). Fixed putting the files in `~/.prusacontrol/data/`

Stevuh, 2017-09-15:
@marcopiraccini that didnt worked for me :/  .
Hope tibor finds a Solution.

MarkFortun, 2017-09-15:
I am also receiving this issue, and I have tried replacing the files as tibor suggested, but am still receiving the issue. 
-Windows 10

tibor-vavra, 2017-09-15:
Can you write me your system version and tree of files in .prusacontrol folder?

Hewster, 2017-09-15:
Got it working, I did this (NOTE: This is on Windows 10):

Downloaded the latest version of PrusaControl(0.9.3.388_beta) from here:
https://prusacontrol.org/#download

Downloaded the latest settings in a zip file from here:
https://github.com/prusa3d/PrusaControl-settings

Pasted the extracted new files here (overwriting existing):
C:\Program Files (x86)\PrusaControl\data

Ran the program, but still not working.

So I then  pasted the new files in:
C:\Users\[**USER NAME**]\.prusacontrol

Ran the program and it worked !

I then uninstalled PrusaControl, and deleted the remaining folders:

C:\Program Files (x86)\PrusaControl
C:\Users\[**USER NAME**]\.prusacontrol

Then installed the latest version again, and it worked straight away :)

So maybe users with this problem simply need to copy the new settings files into their C:\Users\[**USER NAME**]\.prusacontrol folder
or
Uninstall and delete the remaining folders before re-installing.

HTH
Hewster

Stevuh, 2017-09-17:
Hello @tibor-vavra,

I am using Windows 10 Home Edition 64-Bit.
Here are the files in my .prusacontrol Folder.

[Files.zip](https://github.com/prusa3d/PrusaControl/files/1309019/Files.zip)

Stevuh, 2017-09-17:
Hi @ all

I found a Solution and now its working.
I Uninstall Prusa Control and then i searched for "prusacontrol" Folders and deleted all Folders.
Then i run regedit and there i was searching for "prusacontrol" and deleted the Prusacontrol.
Then i restartet my Pc and installed it.

Now it is working.

closed 2017-09-17 by Stevuh

tibor-vavra, 2017-09-18:
Hi @Stevuh,

Iam glad you find solution, today we release new version 0.9.3.390, where this problems are gone. But part of code for updating of printing settings has to be redesigned in future. 

I looked to files(Files.zip), what you attached to your previous post, and inside were .html files named as printing .json settings files. I tought you will download it by download button on github website. Sorry for I didnt tell you that.
[image]