prusa3d/PrusaControl #46: Warning and Print info distorted on Retina windows displays

pacaj2am, 2017-09-21:
Warning and Print info distorted on Retina windows displays.

Probably other items on screen too small (button save gcode, back, buttons to arrange, rotate - 2nd screenshot).

See attached images. Done on Microsoft Surface Pro4 (Screen: 12.3” PixelSense Display , Resolution: 2736 x 1824 (267 PPI))
[image]
[image]

jarnold, 2018-02-25:
[image]

Similar issue on Windows 10 running at 3840 x 2160 running at any scaling percentage.

prusa3d-bb, 2018-02-27:
Hello,
Thank you for your feedback. We will take care of this issue as soon as possible.
Kind regards

—
Lukáš Vacek
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

iandouglas, 2018-03-18:
Yes, it would be nice to get this fixed as well, I can't easily change rotations or placement values on a Retina display:

[image]

prusa3d-bb, 2018-03-21:
Thank you very much, I will forward this info further also.
Kind regards

—
Lukáš Vacek
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

WoodWingStudio, 2018-04-17:
I'm having the same issue.  Windows 10, Microsoft Surface Book, 3000x2000 resolution

[image]