prusa3d/PrusaControl #54: Import Gcode, generate request hangs

sworcester001, 2017-10-31:
If I drag and drop a gcode file (generated in Slic3r) into the program, I can only see a layer view, then hit BACK button, it still shows an active GENERATE button. If I push that it will hang at 11%. I am guessing it shouldn't be active because I already have gcode in the program. If I click the top right X (close) it prompts "Gcode is in generating process Do you want to cancel generating of Gcode and exit
If you don't attempt to close the program, if you wait a bit, the button grays out (not active) and it forgets I ever loaded the gcode. It's gone

If I drag it in there and click SAVE G CODE the counter gets to 99%  and stays. If I click CANCEL nothing happens. If I click the top right X (close) the program closes without any prompts

tibor-vavra, 2017-11-01:
Hi,
yes, generating of empty scene is nonsense. It has to be disabled.

tibor-vavra, 2017-11-23:
Hi, please test it with new version 0.9.4_415