prusa3d/PrusaControl #55: Possible license issue (AGPL vs. GPL)

hroncok, 2017-10-31:
Hi there.

> PrusaControl is an alternative user interface for Slic3r Prusa Edition

Slic3r is licensed with AGPLv3. I'm not sure whether this alternative user interface isn't by design also licensed as AGPLv3, but this repo says GPLv3. Ask you lawyers if needed.

bubnikv, 2017-11-01:
Hmm, this is an interesting topic.

I think the main point of AGPL is: If you use the code as a service, you
have to share it. And that we do indeed.

Vojtech

tibor-vavra, 2017-11-01:
Hi,

I agree with Vojtech. Slic3r PE is placed in folder PrusaControl/tools/Slic3r-Lite in binary form and PrusaControl call his command line interface. Repository of Slic3r PE is https://github.com/prusa3d/Slic3r

Tibor

hroncok, 2017-11-01:
If you agree, why did you close?

tibor-vavra, 2017-11-01:
Im little hot-blooded for this topic :-)))

As I uderstand in section 13 both license AGPLv3 and GPLv3 are compatible for combined work:

Notwithstanding any other provision of this License, you have permission to link or combine any covered work with a work licensed under version 3 of the GNU General Public License into a single combined work, and to convey the resulting work. The terms of this License will continue to apply to the part which is the covered work, but the work with which it is combined will remain governed by version 3 of the GNU General Public License.

hroncok, 2017-11-01:
As I understand it: Combining AGPLv3 with GPLv3 is possible, but the result is de facto AGPLv3.

hroncok, 2017-11-19:
Any news on this topic?

hroncok, 2017-12-10:
Please either tell me your lawyers don't think this is the case or change the license. Ignoring a possible license violation for such a long time bothers me.