prusa3d/PrusaControl #56: Hags at 16%

kateterlecka, 2017-11-01:
Prusa control, version 0.9.3_390_beta slic3r: 1.37.1-prusa3d-win64 hangs at 16% draining over 30% of processor power and heavily writing on disk. Log file is empty, debug mode doesn't help. Reinstalling with removing every single Prusa file didn't help.
The model is a very simple STL - just a box lid.

OS: Win 10 Pro/1703/15063.674
Processor: i75600 2.6GHz
Graphics: Intel HD 5500

kateterlecka, 2017-11-01:
you can find STLs I was trying to slice here: https://drive.google.com/drive/folders/0Bw2572MeU1OBdThMLXRpS3pvRGc?usp=sharing

neutrinus, 2017-11-01:
There are other people experiencing this issue too:
https://shop.prusa3d.com/forum/software-czech-%C4%8Cesky--f9/prusacontrol-t6037.html

tibor-vavra, 2017-11-01:
Hi,
I can confirm this issue. It happened when you have some non-ascii symbol in name of STL file. Slicing engine cant handle it. Next release has workaround for it. At this moment just rename STL file.

Tibor

kateterlecka, 2017-11-01:
Solved! Thanks!

brianhanifin, 2017-11-07:
I am also having this issue. However, my filenames do not have non-ascii symbols in them. They are sans.stl and sans_color.stl.

I am attempting to generate code on my son's account on macOS 10.13.1 and it hangs at 14%.  I can generate gcode on my "Admin" account quickly and without issues. My son's account is "Standard".

tibor-vavra, 2017-11-23:
Hi, please test it with new version 0.9.4_415

ottensmeyer, 2017-12-31:
Not sure, but I believe I'm having the same problem.  Running 0.9.4_415_beta.  If I open a gcode file generated separately (i.e. Prusa3D Slic3r MK2 Multi Material), the path is correctly displayed, however when trying to load an STL (filenames have only ASCII characters), PrusaControl locks immediately after clicking Open.  Any suggestions or additional information that would be necessary for diagnosis?

ottensmeyer, 2017-12-31:
Have tried starting normally, also using Run as administrator, same result

prusa3d-bb, 2017-12-31:
Hello
if you go to prusacontrol.org there is a form there to submit a bug report since that program is in beta.
thank you 

—
Shane
Technical Support

PRUSA Research

+420 222 263 718

+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 7-Holešovice

shop.prusa3D.com

--- original message ---

--- end of original message ---

ottensmeyer, 2017-12-31:
Thanks - have done so.

Is there a prior stable version available for download?  Only the beta seems to be available at  https://prusacontrol.org/#download

Mark

prusa3d-bb, 2018-01-02:
Hello
Prusa Control is still in Beta so there is no offical release or final version of it.
Thank you 

—
ShaneCustomer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

--- original message ---

markfinn, 2018-02-25:
I was stuck at 11% today when I tried to import an STL that was owned by a user different than myself. The file was world readable, but that doesn't seem to be enough.

prusa3d-bb, 2018-02-27:
Hello
on Prusacontrol.org there is a section on that site to submit a bug report and that would go directly to the programmers to look into it. So if you could please fill it out with as much information as possible. 
Thank you 

—
ShaneTechnical Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---