prusa3d/PrusaControl #57: Scale Bugs Out When Changing

DakotaMcCutch, 2017-11-20:
Version: PrusaControl 0.9.4_215

When you change the scale by delete all the numbers and leave the percent sign (X). When you enter a number the other scale fields (Y/Z) will get set to be 9999% even if you only put in 1%(X) in the other fields.

Which causes one hell of a headache trying to fix the scale as well as Prusa Control locks up due to the scale being so large.

prusa3d-bb, 2017-11-20:
Hello,

Is the lock next to the X  Y Z selected? This makes sure you scale evenly along all axes.

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

--- original message ---

--- end of original message ---

DakotaMcCutch, 2017-11-20:
The lock is selected next to the XYZ. I always have to unselect it to fix the scale when it automatically changes the other two to be 99999% thats only if the program doesn't lock up. 

I narrowed  the issue down to be when one of the three is set to be zero in this case I wanted to change them all to be 200% so I deleted the 1 when the lock is enabled it will set the XY to be 99999%.

prusa3d-bb, 2017-11-20:
Hello,

I see the issue now. I will report this. Thank you.

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

--- original message ---