prusa3d/PrusaControl #61: 415 much slower to read and render than on 390 on MacOS

JoshFink, 2017-12-03:
I'm not sure if this effects Windows as well as the Mac but I'm seeing some strange behavior with the new version.

When dragging a gcode file into PC 415 it takes 24m21seconds to read the file and then display and show the estimated time. The same file in 390 beta takes 3 minutes. 

Some more detailed info
2017 MacBook Pro Touchpad
16GB Ram
MacOS 10.3.1

.93_390_beta - Takes 3 minutes to read and render and estimates the total time is 7h 46mins
.93_415_beta - Takes 9m28seconds to read the file to 100% but then doesn't render or display the estimated time until 24m 21seconds. The estimated time is 21h 28m on 415 beta. 

The real actual time to print this on my MK2S was 8h 3min. 

Thanks

prusa3d-bb, 2017-12-04:
Hi there,

I happen to own similar Macbook Pro and I am yet to update. So I will test it out and let you know! 

Best regards

—
Tomáš Chvalina
Technical Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 

shop.prusa3D.com

--- original message ---

--- end of original message ---

tibor-vavra, 2017-12-04:
@JoshFink Hi, I think you mean the MacOS version is 10.13.1

prusa3d-bb, 2017-12-06:
Hi there,

So I have tested out on my Mac and unfortunately for your case I was not able to repeat the problem. Can you maybe try more models? If possible?

—
Tomáš Chvalina
Technical Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 

shop.prusa3D.com

--- original message ---

JoshFink, 2017-12-07:
Hi, thanks for testing it out. I did mean MacOS 10.13.1

As far as the software, I might have been able to fix it. I went into Application Support/PrusaControl and deleted the files there. I then deleted PrusaControl and redownloaded it and moved it into Applications. When I ran it the first time it seemed to have recreated the files and it's much faster now which is strange. Maybe it was a corrupt json file or permissions on the file?  

Either way, thank you for looking into it

prusa3d-bb, 2017-12-12:
Hi there,

That is interesting. I will note it for putting your experience in our knowledge base.

Thank you for the report!

—
Tomáš Chvalina
Technical Support

PRUSA Research
+420 222 263 718 
+421 220 570 305
Partyzánská 188/7A, 170 00 Praha 

shop.prusa3D.com

--- original message ---

DavidSpek, 2017-12-21:
I also have the issue that generating the preview of the G-Code takes extremely long. I tried the above mentioned fix to delete the files in Application Support and reinstall, however it has not solved the problem for me. I also changed the Slic3r version to the one bundled with the new release (1.37.2) and the latest version 1.38.4 which both had no effect. 
Seems that this might be an issue with the graphics rendering (I tested this with a GTX 780 so power should be no issue).

JoshFink, 2017-12-21:
I'm back to having the same problem. PrusaControl is totally unusable for me.

jojowasher, 2017-12-21:
I am having the same problem, have tried reinstalling, still very very slow, just took 30 minutes to slice a plate full of objects, whereas before it would take 2-3 minutes.

jsgraves22, 2018-01-07:
I am having the same issue.  Just upgraded to 415 and generating Gcode seems to be stuck at 100%.  I tried deleting the files mentioned above and reinstalling, but it's a no go.  It hangs at 100%.  I even let it sit for an hour and was never able to save the Gcode. 390 worked perfectly.  The newest version of Slic3r works fine as well.  I'm running a 2016 MacBook pro running Sierra 10.12.6.  Hopefully this gets fixed.