prusa3d/PrusaControl #62: Boot splash screen is forced above all other windows

zbrozek, 2017-12-04:
Using Windows 10, the boot up splash screen hovers over all other windows, effectively rendering the computer unusable until PrusaControl finishes loading. Given that it takes quite some time to load, this is a bit of a nuisance.

It'd be lovely if the splash screen could be sent to the background (e.g., some other window gets focus, alt-tab, etc).

petrzelka, 2017-12-04:
+1 for the fix.
Would it work just to remove "| Qt.WindowStaysOnTopHint"
from
self.splash = QSplashScreen(self.splash_pix, Qt.SplashScreen | Qt.WindowStaysOnTopHint)
in main.py?

tibor-vavra, 2017-12-05:
Yes this is way how to fix it.

iandouglas, 2018-03-18:
Yeah this would be a nice patch.

dr-BEat, 2019-05-19:
+1, blocking everything else on the system is quite annoying. @petrzelka's PR #82 has been open for more than a year already as well.