prusa3d/PrusaControl #63: Failed to execute script main

petrzelka, 2017-12-04:
When I execute PrusaControl.exe (with full path) from another directory it fails with "Failed to execute script main". This fatal error dialog is hidden behind always-on-top splash screen, which must be clicked on to hide and show the error.

I need to run it from another folder because I registered PrusaControl as default app for STL files
.
Also, I don't know why, but according to the code it can open STL file when it's specified in the command line as not first, but second parameter. This works: "PrusaControl.exe blabla myfile.stl" Is it a bug or just missing documentation?

Thanks!

suade907, 2017-12-05:
I'm having the same issue. I had the last version of prusa control working fine and updated to this version. It ran once and now I get the same error as from above. I've tried uninstalling and reinstalling and I still get the error. I'm not doing anything special and am running 64bit windows 10. My prusa control did lock up on a baby groot calculation and I had to force reboot the pc and since I cannot use it.

suade907, 2018-04-23:
I still don't have mine fixed, I believe there is a post somewhere with a fix deleting a file or registry entry after you uninstall, and then reinstall but it didn't work for me. I now have to run the software from a different pc.