prusa3d/PrusaControl #64: PrusaControl 0.9.4_415_beta (Not Responding)

StephanJlx, 2017-12-05:
Hi Guys

I recently became the proud new owner of the MK2 and I am dying to use it but I am having issues with PrusaControl. It opens up etc, but as soon as I bring the part in, big or small, it crashes. It goes into a white none responsive state and I have to force close. I have uninstalled it twice and the fresh install does the same thing. Anybody else having this issue?

Thanks

prusa3d-bb, 2017-12-05:
Hello,

Do you have this issue with Sli3cer Prusa Edition as well?

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

--- original message ---

--- end of original message ---

StephanJlx, 2017-12-05:
Hi

No, I don't have an issue with slic3r prusa edition. It works fine but I
don't like it. Only PrusaControl doesn't work on my PC.

I have managed to get it to work on my slower and older laptop, but this
means I have to work on a cloud based storage to transfer between
computers. Not ideal.

I think the issue might be the graphics cards but the PC actually has a
better CAD graphics card.

Regards

Stephan Le Roux

+420 222 263 718 <+420%20222%20263%20718>

prusa3d-bb, 2017-12-05:
Hello Stephan,

this might be a problem with the software. I'll report it as a bug.

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

--- original message ---

<<https://github.com/prusa3d/PrusaControl/issues/64#issuecomment-349340536>,

lf-, 2017-12-05:
I'm also experiencing this issue on Windows 7 with 0.9.4_415. Where are the log files I can send you so you can find out what the issue is?

StephanJlx, 2017-12-06:
> +420 222 263 718 <+420%20222%20263%20718> <+420%20222%20263%20718>

Thanks!

Don't know if you have a suggestion box? I would like to see PrusaControl
have a gcode editor too. Unless I am missing it? An even better idea is to
build another feature into the Gcode preview toolbar where you can add
color changes. Instead of just being able to add color changes you should
also be able to show where to start or stop printing. That means that you
would have almost flawless print restarts after failures due to power outs
etc. And it can be as simple as picking points on a line with visual
representation.

I am into my second batch of prints and I love my machine. It does crazy
overhangs with zero support...black magic!

prusa3d-bb, 2017-12-06:
Hello,

Since your comment is in here, it should be noted. Thank you very much. We'll see if your suggestions could be used.

We do not implement any gcode editing since we want Prusa Control to be as fool proof as possible.

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

--- original message ---

—

--- end of original message ---

pc4life0427, 2017-12-14:
I am having trouble with newest version of Prusa Control as well...It wont even open on my Mac...it always asks for a Font...different one each time..I have tried to download all over again and 3 times it does the same thing.

tibor-vavra, 2017-12-15:
@pc4life0427 Hi, thank you for your report. Please create new issue with screenshot and log file prusacontrol.log (from your home directory).

Tibor

ottensmeyer, 2017-12-31:
Running 0.9.4_415_beta. If I open a gcode file generated separately (i.e. Prusa3D Slic3r MK2 Multi Material), the path is correctly displayed, however when trying to load an STL (filenames have only ASCII characters), PrusaControl locks immediately after clicking Open. Any suggestions or additional information that would be necessary for diagnosis? Have tried starting normally, also using Run as administrator, same result.  Apologies for multiple postings.

The description of StephaneJlx matches what I observe - Prusa3d Slic3r MK2 Multi Material works fine, but doesn't support 0.1mm layer thickness, hence the need for PrusaControl.

prusacontrol.log is empty (0 byte file)

Thanks, Mark

FredBluggs, 2018-01-07:
I too have the same problem.  

Open PrusaControl, and import an stl file, and PrusaControl becomes non responsive, utilizing 80+% of CPU time.  

Opening the same stl file in Slic3r PE and behaviour is as expected, without issue.

Windows 10 Pro ver 1703
Slic3r PE version 1.38.6-prusa3d-win64
PrusaControl 0.9.4_415_beta.
STL file used is castle.stl from C:\Users\Public\Documents\Prusa3D\Objects 213\MK3
prusacontrol.log is a zero byte file.

Regards,
Fred

neilgroulx, 2018-01-17:
I'm also having this same problem. Files open up fine in Slic3r Prusa edition.
PrusaControl starts up OK, but as soon as I load a file it becomes non-responsive and I have to force kill it. Completely unusable!

lonepinetech, 2018-03-18:
I was using Prusa Control with no issues for many prints. I then upgraded my GPU from the integrated gpu  on my gigabyte h27dm-ds3h motherboard to a quadro k620 dedicated gpu.  Prusa Control now goes not responsive upon import of an .stl. I will miss it!

payala, 2018-05-25:
I had the same issue, I also have an nVidia Quadro graphics card, and the workaround of @suremicro worked perfectly for me.

lf-, 2018-05-26:
@suremicro Interesting! I was reproducing this issue with a Quadro on Win7.

VHCEngineering, 2018-07-25:
Any news on this? I too am having an issue on Windows 7 with quadro Graphics card.

lf-, 2018-07-25:
Check your driver is up to date. I know on the machine I reproduced it on
had a prehistoric driver installed.

VHCEngineering, 2018-07-25:
The graphics card driver is up to date

beeb, 2018-08-03:
Can reproduce the issue on my computer with Windows 10 and a Nvidia Quadro graphics card. Program becomes unresponsive upon loading of STL.

throb, 2018-08-26:
Confirmed here also with Quadro 5000 on Windows 10 with latest NVidia Drivers.
Win8 Compatibility mode also solves the issue.