prusa3d/PrusaControl #66: Allow truely solid prints

mrjohnc, 2017-12-19:
In the infill settings it describes 70% as solid, however it would be extremely useful to offer the option for the whole shape to be printed solid. In my case for when printing with transparent filaments so you can't see infill pattern.

prusa3d-bb, 2017-12-20:
Hello John,

What slicer are you using? In PE Slic3r you can set any infill percentge in the Print settings tab.

Regards,

—
Edmund Trubáček
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

--- original message ---

--- end of original message ---

mrjohnc, 2017-12-22:
Thanks, I'm using PrusaControl to do everything, I haven't changed any settings, just trying to change the infill to be 100% (by which I mean a solid layer) and it only allows 70% maximum

prusa3d-bb, 2017-12-27:
Hello,

I am afraid that 100% infill is not possible in PrusaControl. You can use Slic3r to create gcode with 100% infill.

Regards,

—
Edmund Trubáček
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

--- original message ---

mrjohnc, 2017-12-27:
Yes, I know, that's why I'm filling this feature request.

One way to achieve this would be to have an option in PrusaControl called 100% infill but in reality it just changed the number of solid bottom layers to 9999 or something.

Thanks

Moorviper, 2018-04-14:
you can edit the json files to get 100% infill

mrjohnc, 2018-04-14:
@Moorviper that's a nice option for programmers, but it would be nice if muggles like me were able to do this too

plorimer, 2018-04-15:
I agree that this would be a great addition. I'm trying to print lithophanes generated from 3dp.rocks, which slic3r prusa edition for some reason refuses to slice, but Prusa Control has no problems with. However, the illusion made by a lithophane requires actual solid infill as the density if what causes the varying greytones.

squatsandscience, 2018-10-05:
I would like this as well!

MarcusAdams, 2018-12-23:
Prusa MK3 create less than satisfactory lithophanes anyway due to inconsistent extrusion. However, I use Cura for most of my slicing anyway. I only use Slic3r when I'm using the MMU2.