prusa3d/PrusaControl #70: PrusaControl not exporting .gcode from Slic3r.

Tucker2015, 2017-12-29:
If I slice a .STL in Slic3r then export the .gcode and import into PrusaControl to do colour changes when I export the .gcode PrusaControl just hangs and does not complete the export.

But if I slice a .STL from Simplify3D and try the colour change it exports fine.

Ive tried this on both a Mac and Windows.

Slic3r version 1.38.5-prusa3D
PrusaControl 0.9.4_415_beta

prusa3d-bb, 2017-12-29:
Hello,

I would suggest using this website for already sliced files: https://www.prusaprinters.org/color-print/

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

For simple troubleshooting help you can check our help page: http://help.prusa3d.com

--- original message ---

--- end of original message ---

Tucker2015, 2017-12-29:
Thanks Jennes, I have been using this today and it works fine.  Just thought I would let you know about the issue.

prusa3d-bb, 2017-12-29:
Hello,

Thank you very much for letting us know.

Best regards,

Jennes De Schutter
Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

Partyzánská 188/7a, 17000, Prague

shop.prusa3D.com

For simple troubleshooting help you can check our help page: http://help.prusa3d.com

--- original message ---

FJC2, 2018-03-24:
I have the exact same problem.  Upon clicking the "save g-code" button it will hang before getting to 100%.  Most commonly for me, it'll hang at 98%.  The resulting output file will have the M600 commands, but will be missing a few hundred lines of the rest of the gcode.

[plate1.zip](https://github.com/prusa3d/PrusaControl/files/1844362/plate1.zip)

Tucker2015, 2018-03-24:
Guessing it’s not been fixed then in latest version ?

FJC2, 2018-03-24:
Sorry I should have specified - I'm on 0.9.4_415, which I think is still the latest version as of this date.