prusa3d/PrusaControl #71: Generating G-code preview hangs or takes incredible long time

ImreZol, 2018-01-02:
I try to slice the Track.stl from https://www.thingiverse.com/thing:1385312 with default settings.

After slicing, the G-code generration hangs.
Prusa Control is running and responding (I can cancel the operation). It's use ~90% CPU (I thing it means  use 1 core). I gave up after an hour...

With Slic3r Prusa Editon 1.38.6-prusa3d it (slicing and G-code export) succeed within 1-2 minutes (with default settings).

Prusa Control Ver0.9.4_415_beta (Slic3r engine ver:1.37.2.1-prusa3d)
Printer: mk2s
Material: Prusa PLA
Quality: Optimal -0.15 mm
Infill: Standard - 20%
Support: None
Brim: Off

Os Ver: MacOS High Sierra 10.13.2
Computer: MacBook Pro(115-inch,2016) 
Processor 2,6 Ghz Intel Core i7
Memory: 16 GB

jsgraves22, 2018-01-07:
Same issue for me with 415.  Similar MacBook.  G-code generation hangs at 100% and I'm never able to save it.  I have waited nearly an hour and it drains my MacBooks battery like a champ but never generates.  Slic3r works fine.

cdburgess, 2018-01-24:
I am experiencing the same issues. So I upgraded to HighSierra 10.13.3 and I am still getting the same issues. I do think it's hit and miss. One of the jobs I was working on finally shows the `Save G-Code` button. But it was quite a while after it hit 100%.

Prusa Control Ver0.9.4_415_beta (Slic3r engine ver:1.37.2.1-prusa3d)
Printer: MK3
Material: Prusa PLA
Quality: Optimal -0.15 mm
Infill: Standard - 20%
Support: None
Brim: Off

Os Ver: MacOS High Sierra 10.13.3
Computer: Mac mini (Late 2012)
Processor 2.3 GHz Intel Core i7
Memory: 16 GB 1600 MHz DDR3

cdburgess, 2018-01-25:
I did discover that if there are possible issues with the model, it will never finish. I have some small objects I am trying to print multiples of. When I only do a few, the g-code button will appear. But I can tell there's a problem with the model because not all the layers are there when I drag the layer viewer up and down. After repairing the model, it will eventually finish and I can export the g-code.

cknowles2112, 2018-02-17:
doing some more research - even with known good models, things can take an insanely long time.  But, if I scale the model down to 20% of size, it takes MUCH less time.  All of this is in comparison to S3D's timing.  I think there's a correlation between volume of the model, and the time that the g-code preview generation takes.

joelkoz, 2018-03-05:
Just happened to me also.  Macbook Pro, 16 GB RAM, High Sierra 10.13.3

ldestrooper, 2018-03-18:
Also just had this issue with PrusaControl 0.9.4_415_beta on macOS High Sierra 10.13.3

magnumuk, 2018-03-24:
This has been a problem for several months now, I wonder why no fix has been issued yet?

leewrigh7, 2018-03-30:
Same problem here,  attempting to print thing:19104 (thingverse)...

Stuck at 100%, therefore can't save to SD or print.

2014 Macbook pro, 16GB, High Sierra 10.13.3, Latest PrusaControl (as of 30th March 2018).

Prusa PLA, 0.15mm, Solid, Support: none, Brim:off

knot2nice, 2018-04-08:
still a problem for me as well...sigh

maximlevitsky, 2018-04-08:
Same here on Linux. Seems to work fast enough in Win10 on same machine. Didn't test rigoursly though.
Maybe this is somehow OpenGL related?

JoschaMiddendorf, 2018-04-15:
im having the same problem

magnumuk, 2018-04-15:
It looks like the only platform that is being supported is Windows, nothing for Mac and Linux users.

buffedelic, 2018-04-27:
mee too, MacOS 10.13.4 and latest PrusaControl. Smaller objects slices with no significant problem but larger more complex models just keep spinning until I run cmd-q and fire upp slic3r

hosek, 2018-05-03:
I have same issue, but regarding the status bar, problem is probably not in generation gcode itself, but generating G-code preview. Just thought - is there some condition, that save gcode button is shown only after preview is ready? Then if preview somehow crash or take too long for complex objects, user will never see the saving button even the gcode file is ready.

magnumuk, 2018-05-03:
This problem has been around for some time now.  I would have expected a fix by now! I wonder if anyone is actually working on the problem?

Ahbrown41, 2018-05-04:
Same issue, S3D is so fast, this is terribly slow.

workinghard, 2018-05-10:
Same observation here. MacOS most recent version. gcode is getting generated but the preview of the gcode is not possible for more complex objects. Because you can't save it, PrusaControl can't be used this way at all :(

alhazar313, 2018-05-12:
I have the same issue, the status bar gets right near the end and stops pegging the cpu @90%, its concerning that it seems this issue isn't being addressed or acknowledged....

frenchrabbit, 2018-05-14:
Same. For some stl if I wait for 20minutes it gives preview and allowing to save , sometimes it never appears. So every print preparation is like lottery.

gavztheouch, 2018-05-22:
just tried again and like another member it looks like scaling down the models to say 20%  seems to work. Maybe its the code that creates an estimate time for the g-code to run as slic3er does not have this feature and also works ok.

workinghard, 2018-05-22:
@hosek : Printing a gcode without checking it, is not an option! Of course you could use any online tools for checking but this miss the point of PrusaControl completely. It supposed to be simple, so my wife can load, preview and print. If that doesn't work, it missed his target and will not be used. I see this bug as pretty urgent and major for this project. But it looks like there are no resources available right now

buffedelic, 2018-05-22:
I agree with @hosek, preview of gcode is cruzial to determine if print will be good or not. Especially when using supports.

The ease of use whas really appealing to me when learning my kids the basics. Now they had to learn the hard way :) lots of profiles made for various situations

hosek, 2018-05-23:
@workinghard yes, I agree with you. It wasn't supposed to be solution it's just temporary workaround for people who want to use easy UI of Prusa Control, but are able to check gcode somewhere else... just to know that it's possible. Sorry if it was not clear from my post.

joelkoz, 2018-06-09:
Update: This is definitely an issue with SLOW gcode preview generation, and not locking up entirely.  I had a very large model that seemed to "get stuck" on gcode preview generation. It was late at night and I gave up after 30 minutes of waiting, but I did NOT stop the process and I just went to bed.  When I came back in the morning, the preview was available and the save worked as normal.

WarrenSchultz, 2018-08-08:
I have a question on this, as I'm also experiencing this issue. I've tried exporting the gcode locally as well as sending direct to the printer. (Octoprint on RPi 3B+)

The final gcode generation does not seem to be multithreaded, or if it is, it isn't utilizing all cores. 
How is generation currently handled? Serially by object, or by layer? (or?)

Would it be possible to create volumes around each object on the plate, and then launch separate threads for code generation per object before re-merging the code? (I understand this may be a naive request, since I don't know how that code works under the hood).

AnatolLocker, 2018-08-14:
Looks like this problem isn’t high on the developers list...

ImreZol, 2018-08-14:
Because Slic3r PE will become the PrusaControl 2 (https://www.prusaprinters.org/prusa-research-summer-update-2018) I close this issue...

MartinMajewski, 2018-11-23:
How can you close this issue if it is a nagging problem for all slic3r PE users on a Mac?! I use the most current version (1.41.2-rc+) and have had this issues for months now. So it is a problem with the current development branch of slic3r. I pledge for reopening this issue and really have a look at it. Exporting gcode / sending gcode to OctoPrint is no fun at all at the moment!

ImreZol, 2018-11-24:
Hi Martin,

This issue was about PrusaControl not Slic3r.
I closed this issue, because prusa no longer develop this project (the last commit was  on Jan 5, 2018).

I am not a contributor just a simple user, but I think if you have this problem with Slic3r, you should open a ticket in Slic3r project repo: 
- Slic3r -> Help -> Report an Issue 
- or  use this link: https://github.com/prusa3d/slic3r/issues/new

I do not have this problem with Slic3r PE 1.41.2+, so you should try this version at first.
(I have 10.14.1 Mojave on my Mac)

closed 2018-11-24 by ImreZol

psavva, 2020-05-17:
Unchecking "Avoid Crossing Parameters" makes the lag go away.

handell, 2020-05-20:
The new PrusaSlicer fixed the issue. THANKS!