prusa3d/PrusaControl #72: Windows 7 - crash

jfojtik, 2018-01-04:
I use Windows 7 64bit, always when I open 3D model in PrusaControl (0.9.4_415_beta) so PrusaControl [freezes](http://take.ms/9IBNC). I tried another computer with Win 7 and there was the same problem. SW works only Windows 10. Why?

tibor-vavra, 2018-01-04:
Hi,
when you start PrusaControl is it function? Can you see heatbed and printing space, widgets in scene and on right side? And when you drop some .stl file its freez?

jfojtik, 2018-01-04:
Hu, yes, when I start PrusaControl al lis ok. When I load .stl, PrusaControl its freez.

J. Fojtík

Zdravím, ano při startu je PrusaControl plně funkční, zamrzne to až když nahraju .stl.

J. Fojtík

Myslete na životní prostředí dřív, než zbytečně vytisknete tento e-mail.
Please think about the environment before printing out this e-mail.
Informace obsazene v tomto e-mailu jsou urceny pouze a vylucne pro potreby jeho adresata. Text nebo prilohy mohou obsahovat utajovane informace, informace povazovane spolecnosti MORAVIA PROPAG, s.r.o. za obchodni tajemstvi, pripadne jine informace podlehajici ochrane dle prislusnych pravnich predpisu. Pokud Vam byl tento e-mail dorucen omylem, zdrzte se, prosim, jakekoli manipulace s textem ci prilohami, jako je kopirovani, presmerovani, zpristupneni treti osobe a podobne. O chybnem doruceni informujte odesilatele a e-mail vcetne priloh nenavratne vymazte ze sveho pocitace.
The information contained within this e-mail is intended only and exclusively for the person or entity to which it is addressed. The text or attachments may contain confidential information, information considered a trade secret by MORAVIA PROPAG, s.r.o. or, as the case may be, other information subject to protection under the relevant legal regulations. If you receive this e-mail by mistake, please refrain from copying, forwarding or disclosing the text or attachments to other persons, etc. Inform the sender of the mistaken delivery and beyond delete the e-mail, including all attachments, from your computer.

malkub, 2018-01-04:
Hello i´v got same problem in windows 10 pro

stoerenfried, 2018-01-10:
Win10 pro, two PCs with i7 / i5, same problems. Seems to me that some background check or somethnig starts and breaks, but I could (so far) not influence this by changing settings.

ndee85, 2018-04-17:
I have also noticed crashes. I noticed when exporting stl files from blender, Prusa Control crashes trying to load that stl files. Could it have something todo with the stl parser?

VHCEngineering, 2018-07-25:
Same here. is this a graphics card issue? I am on a quadro