prusa3d/PrusaControl #73: Allow users to set default quality in Settings

mrjohnc, 2018-01-16:
I really love PrusaControl but I print almost always in draft quality (I'm working on an open hardware project where there are lots of prototype prints). It is a pain to realise an hour into a print you forgot to change from 0.15mm layer height to 0.3mm layer height and your print is going to take 6 hours not 3 hours.

Thanks

prusa3d-bb, 2018-01-22:
Hello, 

thank you for the feedback, our developers will get back to you soon.

Best regards,

Honza Vysušil
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

--- original message ---

--- end of original message ---

vitriolix, 2018-03-13:
👍

iandouglas, 2018-03-18:
There's a draft mode in the filament settings, but the software doesn't allow us to access it.