prusa3d/PrusaControl #76: Feature request: checkbox for raft

ccfoo242, 2018-02-12:
I would like to be able to use rafts and not just brim.

prusa3d-bb, 2018-02-14:
Hello Steve,

The raft option is on Slic3r Prusa Edition too.

Regards,

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

ccfoo242, 2018-02-14:
Yes, I am able to use that, but PrusaControl has an easier/nicer interface so it would be nice to have a raft option next to the brim option.

prusa3d-bb, 2018-02-16:
Hello,

We are working in Prusa Control and will be improved with new features in the future.

Have a nice weekend!

—
Blas Palencia
Customer Support

PRUSA Research
+420 222 263 718
+421 220 570 305
188/7a Partyzánská, 17000, Prague
shop.prusa3D.com

Please rate our reply ...be generous :-)

--- original message ---

vitriolix, 2018-03-13:
👍

cwei, 2018-03-13:
👍

Viele Grüße
Carsten

crua9, 2018-07-16:
This is a needed feature