prusa3d/PrusaControl #87: Infill is shit with MK3

swissgirl, 2018-03-28:
If I choose the printer MK3 in the settings the infill is rectangular and the crossing points are always the same. Therefore the crossing points are higher than the rest of the print and the nozzle crashes into the printed part.

If MK2 is chosen the infill is 120° and different for each layer.

[image]
[image]

frenchrabbit, 2018-05-14:
Same issue. Has to keep MK2 profile for my MK3

gaellafond, 2018-06-03:
I just realised that today. I prefer MK2's infill. Why is it different depending on the printer?