prusa3d/PrusaControl #88: Trouble installing or running prusacontrol and/or slic3r PE

nahtebee, 2018-04-10:
Sorry for the long post. Mostly it's to show that I did try what I thought were the most logical things, but had little success.

I'm trying to install and/or run either prusacontrol or slic3r PE, but either I'm dumber than I think (possible) or something is wrong.
I'm on Kubuntu 17.10 (and will upgrade to 18.04 within a week).

I'm new to 3d printing so logically I attempted to download prusacontrol_0.9.4.414_beta.AppImage. 
(Suggestion: It would be nice to add the simple chmod instruction somewhere obvious for those of us not familiar with AppImage or Snaps or whatever the next cross-platform app-executor-du-jour is. Or just maybe publish a ppa and make it easier :-D ). Spent time thinking it was a direct executable...then a python image (since that was mentioned somewhere) then I pulled out Google like I should have at first.

So after the chmod a+e *.AppImage to make the AppImage executable, I get:
```
nahtebee@jaffa:~/Downloads$ ./prusacontrol_0.9.4.415_beta.AppImage 
/bin/bash: /tmp/.mount_prusacohkx43/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
desktop-file-validate is missing. Skipping /tmp/.mount_prusacohkx43/usr/bin//prusacontrol.wrapper.
/bin/bash: /tmp/.mount_prusacohkx43/lib/x86_64-linux-gnu/libtinfo.so.5: no version information available (required by /bin/bash)
Dpi coef: 0.9895833333333334
Dpi scale: 2
Sirka praveho panelu: 247
GUI set normal support settings
Segmentation fault (core dumped)
```

So maybe I needed slic3r PE. Seems that was mentioned, though not definitively in simple steps. I looked in github, prusa3d forums and knowledgebase, then google. 
So this time I was smart and downloaded the tar.bz, since I'm familiar with what to do with that. Extracted, cd to the new directory, tried running ./slic3r, got the command line help. Hmm. 
No install script (even though github or somewhere mentioned a do-install script) and no Readme file or anything to know what to do.

So tried running ./slic3r --gui, with poor results:
```
nahtebee@jaffa:~/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854$ ./slic3r --gui

(slic3r.pl:3836): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:3836): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
Can't load '/home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/auto/OpenGL/OpenGL.so' for module OpenGL: libglut.so.3: cannot open shared object file: No such file or directory at /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/DynaLoader.pm line 193.
 at /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/Slic3r/GUI/Plater/2DToolpaths.pm line 134.
Compilation failed in require at /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/Slic3r/GUI/Plater/2DToolpaths.pm line 134.
BEGIN failed--compilation aborted at /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/Slic3r/GUI/Plater/2DToolpaths.pm line 134.
Compilation failed in require at /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/Slic3r/GUI.pm line 21.
BEGIN failed--compilation aborted at /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib/site_perl/5.22.0/Slic3r/GUI.pm line 21.
Compilation failed in require at (eval 77) line 1.
```

Also tried executing slic3r from Dolphin by double click, with no success, as expected.

Tried running ./slic3r.pl, with expectedly poor results as well.
```
nahtebee@jaffa:~/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854$ ./slic3r.pl 
Can't locate local/lib.pm in @INC (you may need to install the local::lib module) (@INC contains: /home/nahtebee/Downloads/Slic3r-1.39.1-prusa3d-linux64-full-201803010854/lib /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.0 /usr/local/share/perl/5.26.0 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at ./slic3r.pl line 9.
BEGIN failed--compilation aborted at ./slic3r.pl line 9.
```

Well, maybe the slic3r AppImage will behave better. Downloaded it, chmod, and execute:
```
nahtebee@jaffa:~/Downloads$ chmod a+x Slic3r-1.39.1-prusa3d-linux64-full-201803010854.AppImage 
nahtebee@jaffa:~/Downloads$ ./Slic3r-1.39.1-prusa3d-linux64-full-201803010854.AppImage 
^C^C^C^C
```
...it hung, or at least the gui never showed its face. Too embarrassed by my poor attempts?

Just for grins I also downloaded the standard Slic3r-master-latest.AppImage, and surprisingly, it ran, after a few warnings, the main window appeared. (Didn't try to use it).
```
nahtebee@jaffa:~/Downloads$ ./Slic3r-master-latest.AppImage 

(slic3r.pl:4883): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:4883): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
/usr/lib/x86_64-linux-gnu/libproxy.so.1: symbol _ZTTNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEE, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so
```

Can someone point out to me the dumb mistake I'm making that everyone else obviously knows about?
Thanks for any help!

probonopd, 2018-04-10:
> Suggestion: It would be nice to add the simple chmod instruction somewhere obvious for those of us not familiar with AppImage

Agree, that would be useful for users new to AppImage.

> Or just maybe publish a ppa and make it easier :-D

ppas don't work for distributions that are not Debian/Ubuntu based.
Do you really think it can be easier than this?

[image]

> Spent time thinking it was a direct executable...

It is "direct executable" but you need to set the executable bit, this is how security on Unix works (and has worked for the past decades). It's a feature, not a bug ;-)

`prusacontrol_0.9.4.414_beta.AppImage` and `Slic3r-1.39.1-prusa3d-linux64-full-201803010854` both are working fine for me on Xubuntu 16.04.

> Slic3r-master-latest.AppImage 

That one is not from Prusa, and it gives me this error:

```
me@host:~/Downloads/QuizTech.AppDir/usr$ /home/me/Downloads/Slic3r-master-latest.AppImage
libGL error: unable to load driver: r600_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: r600
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 1725 error_code 2 request_code 155 minor_code 24)
```

probonopd, 2018-04-10:
Fwiw, `PrusaControl_0.9.4_415_beta_64bit.AppImage` is running fine for me on `ubuntu-18.04-beta2-desktop-amd64.iso`:

[image]

nahtebee, 2018-04-10:
Thanks for the reply!

> Suggestion: It would be nice to add the simple chmod instruction somewhere
> obvious for those of us not familiar with AppImage
>
> Agree, that would be useful for users new to AppImage.
>
> Or just maybe [...]

Haha...no. Just for familiar.

<https://camo.githubusercontent.com/12596dace908e76456b9bb24906940f91c82cdff/68747470733a2f2f646973636f757273652d63646e2d736a63322e636f6d2f7374616e6461726431322f75706c6f6164732f617070696d6167652f6f7074696d697a65642f31582f613438383963356362383731316436383435623538313335303830373837643266333730616633355f315f353030783530302e676966>
>
> Spent time thinking it was a direct executable...
>
> It is "direct executable" but you need to set the executable bit, this is
> how security on Unix works (and has worked for the past decades). It's a
> feature, not a bug ;-)
Yes...it was just my problem, not knowing a thing about the AppImage format
and what to do with it. And, I suppose, nothing documenting what to do with
it, assuming everyone "knows". Like I say, I finally Googled it out and got
it going.

> prusacontrol_0.9.4.414_beta.AppImage and Slic3r-1.39.1-prusa3d-linux64-
> full-201803010854 both are working fine for me on Xubuntu 16.04.
Ok. Maybe I got a bad download of prusacontrol. I'll try downloading again.

In a weird twist, as I tried to shut down my computer, the Slic3r PE
interface popped up! So it was working, just that my 17.10 wasn't
displaying it (and I did look for it, on all screens). So I'll try that
again tonight.

> Slic3r-master-latest.AppImage
>
> That one is not from Prusa, and it gives me this error:
Um...yeah. Sorry, I thought I had corrected all those locations. This is
the original Slic3r, not PE. But interestingly enough, it ran for me.
But I want to use the PE edition, for obvious reasons.

me@host:~/Downloads/QuizTech.AppDir/usr$
/home/me/Downloads/Slic3r-master-latest.AppImage
> libGL error: unable to load driver: r600_dri.so
> libGL error: driver pointer missing
> libGL error: failed to load driver: r600
> libGL error: unable to load driver: swrast_dri.so
> libGL error: failed to [...]
Thanks for your reply. If my testing tonight doesn't work, then I'll post
another question, but I appreciate your help.

Ethan

nahtebee, 2018-04-10:
> Fwiw, PrusaControl_0.9.4_415_beta_64bit.AppImage is running fine for me
> on ubuntu-18.04-beta2-desktop-amd64.iso:
Very cool! I wasn't quite bold enough to run the beta. But maybe I need to!

Ethan