prusa3d/PrusaControl #91: PrusaControl 0.9.4_415_beta - missing geometry

Drifter4242, 2018-04-22:
Slicing this geometry only results in half of it being sliced.

The geometry is somewhat complex, so I suspect something overflowed.

I've attached the stl and screengrab of the result.

[grill-r.zip](https://github.com/prusa3d/PrusaControl/files/1936092/grill-r.zip)

[image]

prusa3d-bb, 2018-04-24:
Hello,

The Slic3r app detected 8 backwards edges and it tries to fix. Seems able to go over the issue and let you slice.
PrusaControl instead cannot sort the issue.

You may find this article interesting https://www.prusaprinters.org/repair-3d-models-errors/.

Regards

—

Michele Moramarco

Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---

--- end of original message ---

VHCEngineering, 2018-07-25:
Had a similar issue with text emboss. Was supposed to say JESUS. Printed JES.

prusa3d-bb, 2018-07-30:
Hello,

Please give it a try selecting  "detect thin walls" in our new Slic3r PE in the first "print settings" tab.
Even better, print with a 0.25 mm nozzle! 
You can find some more information in details about this at https://www.prusaprinters.org/everything-about-nozzles-with-a-different-diameter/.

Regards

—

Michele Moramarco

Customer Support

PRUSA Research

+420 222 263 718

+421 220 570 305

188/7a Partyzánská, 17000, Prague

shop.prusa3D.com

For simple troubleshooting, please use: http://help.prusa3d.com

Please rate our reply ...be generous :-)

--- original message ---