prusa3d/PrusaControl #93: AppImage packaging is either too leaky or not leaky enough

CounterPillow, 2018-05-09:
On Arch Linux:

```
$ ./prusacontrol_0.9.4.415_beta.AppImage 
Traceback (most recent call last):
  File "./PrusaControl-master/main.py", line 10, in <module>
  File "./PrusaControl-master/controller.py", line 21, in <module>
ImportError: /usr/lib/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var
$ objdump -T /usr/lib/libfontconfig.so.1 | grep FT_Done_MM_Var
0000000000000000      DF *UND*  0000000000000000              FT_Done_MM_Var
```

Either include all of your dependencies or none of them.

probonopd, 2018-05-09:
Specifically, either bundle both of or none of Harfbuzz and FreeType. At least this seems to be the conclusion from earlier discussions. In the https://github.com/AppImage/AppImages/ project we no longer bundle libharfbuzz and libfreetype as per https://github.com/AppImage/AppImages/commit/9082bbe3b924b656d8675da64a69c6b2b32ac7b4

References:
* https://github.com/AppImage/AppImages/pull/323
* https://github.com/probonopd/linuxdeployqt/issues/261#issuecomment-377522251
* https://github.com/probonopd/linuxdeployqt/issues/157#issuecomment-320755694

tracernz, 2018-06-25:
Also hitting this.
```
$ ./prusacontrol_0.9.4.415_beta.AppImage
Traceback (most recent call last):
  File "./PrusaControl-master/main.py", line 10, in <module>
  File "./PrusaControl-master/controller.py", line 21, in <module>
ImportError: /usr/lib/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var
$ pacman -Q fontconfig
fontconfig 2.13.0+10+g58f5285-1
$ objdump -T /usr/lib/libfontconfig.so.1 | grep FT_Done_MM_Var
0000000000000000      DF *UND*	0000000000000000              FT_Done_MM_Var
$ pacman -Q freetype2
freetype2 2.9.1-1
$ objdump -T /usr/lib/libfreetype.so | grep FT_Done_MM_Var
000000000001c150 g    DF .text	0000000000000026  Base        FT_Done_MM_Var
```

buddyabaddon, 2018-09-02:
I am also hitting this error :(

$ ./prusacontrol_0.9.4.415_beta.AppImage 
Traceback (most recent call last):
  File "./PrusaControl-master/main.py", line 10, in <module>
  File "./PrusaControl-master/controller.py", line 21, in <module>
ImportError: /usr/lib64/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var
$ equery l fontconfig
 * Searching for fontconfig ...
[IP-] [  ] media-libs/fontconfig-2.13.0-r4:1.0
$ equery l freetype
 * Searching for freetype ...
[IP-] [  ] media-libs/freetype-2.9.1-r3:2

probonopd, 2018-10-20:
@castaway you could extract the AppImage, remove the libraries, and try if the application runs then:

```
# Download the AppImage
wget -c "https://github.com/prusa3d/PrusaControl/releases/download/v0.9.4/PrusaControl_0.9.4_415_beta_64bit.AppImage"

# Make the AppImage executable
chmod +x PrusaControl_0.9.4_415_beta_64bit.AppImage

# Extract the contents of the AppImage
./PrusaControl_0.9.4_415_beta_64bit.AppImage --appimage-extract

# Delete the library in question
find squashfs-root/ -iname '*freetype*' -delete

# Run the extracted AppImage to see if the error is resolved
./squashfs-root/AppRun
```

castaway, 2018-10-21:
> ./squashfs-root/AppRun

Aha! Done, and yes that now runs just fine, thanks!

Can I repack it easily from this state?

probonopd, 2018-10-21:
Easy:

```
# Get appimagetool
wget -c "https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage"

# Make it executable
chmod +x appimagetool-x86_64.AppImage 

# Convert AppDir into AppImage
./appimagetool-x86_64.AppImage squashfs-root/

# Test whether it works
./PrusaControl-x86_64.AppImage 

```

castaway, 2018-10-21:
> wget -c "https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage"

AppImage of the AppImage tool, of course!

tak-21, 2019-02-24:
@probonopd: thx!!!
One more question please: Under Ubuntu 18.10 with gnome I can right-click the app and add it as a favourit. When I click this icon which was added to the bar, the program starts and/but there is another icon showing up which represents the now running program. This behaviour is different when I start e.g. firefox. It is just cosmetics but can this be "fixed"?

probonopd, 2019-02-24:
You need to launch it through a desktop file. Tools like the optional [appimaged](https://github.com/AppImage/appimaged) daemon or [AppImageLauncher](https://github.com/TheAssassin/AppImageLauncher) can do this for you automatically.