prusa3d/PrusaControl #108: Segmentation fault (core dumped)

xqgex, 2018-07-28:
I am getting a seg fault error when running PrusaControl,
File name: prusacontrol_0.9.4.415_beta.AppImage
Downloaded from: https://prusacontrol.org/#download
System version: Fedora (4.17.6-200.fc28.x86_64)

probonopd, 2018-07-28:
Please run `LD_DEBUG=libs ./prusacontrol_0.9.4.415_beta.AppImage`, do you see any errors?

xqgex, 2018-07-28:
The output is 5591 lines long,
Doing search for the words "error" and 'fatal" gave me:

15 times (Lines 2711 until 2725):
`     32326:	/lib64/libGL.so.1: error: symbol lookup error: undefined symbol: FunctionType (fatal)`
90 times (Lines 2765 until 2854):
`     32326:	/lib64/libGLU.so.1: error: symbol lookup error: undefined symbol: FunctionType (fatal)`
lines 3620 to 3638:
```
     32326:	find library=libgpg-error.so.0 [0]; searching
     32326:	 search path=/tmp/.mount_prusac5QfCZ6/usr/lib:/tmp/.mount_prusac5QfCZ6/usr/lib/x86_64-linux-gnu:/tmp/.mount_prusac5QfCZ6/lib:/tmp/.mount_prusac5QfCZ6/lib/x86_64-linux-gnu:tls/x86_64/x86_64:tls/x86_64:tls/x86_64:tls:x86_64/x86_64:x86_64:x86_64:		(LD_LIBRARY_PATH)
     32326:	  trying file=/tmp/.mount_prusac5QfCZ6/usr/lib/libgpg-error.so.0
     32326:	  trying file=/tmp/.mount_prusac5QfCZ6/usr/lib/x86_64-linux-gnu/libgpg-error.so.0
     32326:	  trying file=/tmp/.mount_prusac5QfCZ6/lib/libgpg-error.so.0
     32326:	  trying file=/tmp/.mount_prusac5QfCZ6/lib/x86_64-linux-gnu/libgpg-error.so.0
     32326:	  trying file=tls/x86_64/x86_64/libgpg-error.so.0
     32326:	  trying file=tls/x86_64/libgpg-error.so.0
     32326:	  trying file=tls/x86_64/libgpg-error.so.0
     32326:	  trying file=tls/libgpg-error.so.0
     32326:	  trying file=x86_64/x86_64/libgpg-error.so.0
     32326:	  trying file=x86_64/libgpg-error.so.0
     32326:	  trying file=x86_64/libgpg-error.so.0
     32326:	  trying file=libgpg-error.so.0
     32326:	 search cache=/etc/ld.so.cache
     32326:	  trying file=/lib64/libgpg-error.so.0
     32326:	
     32326:	
     32326:	calling init: /lib64/libgpg-error.so.0
```
Lines 4008, 4479 and 5308:
`     32326:	/lib64/libharfbuzz.so.0: error: symbol lookup error: undefined symbol: FT_Get_Var_Blend_Coordinates (fatal)`

probonopd, 2018-07-28:
Yes, [HarfBuzz strikes again](https://gitlab.com/probono/platformissues#harfbuzz)! It seems to be so closely linked to FreeType that either both or none must be bundledm but not one of them without the other.

Please try whether this fixes it:

```
./prusacontrol_0.9.4.415_beta.AppImage --appimage-extract
find squashfs-root/ -name *eetyp* -delete
./squashfs-root/AppRun 
# Does it run?
```

xqgex, 2018-07-28:
Running that command created a new folder called "squashfs-root", I've try to do
 ```
cd squashfs-root/
./AppRun
```
But still got "Segmentation fault (core dumped)"

xqgex, 2018-07-28:
Based on:
https://gist.github.com/cryzed/e002e7057435f02cc7894b9e748c5671
It's seems like a problem with the freetype version

xqgex, 2018-07-28:
My bad, I thought that "find" is an arguments for the AppImage
Doing:
1. ./prusacontrol_0.9.4.415_beta.AppImage --appimage-extract
2. find squashfs-root/ -name *freetype* -delete
3. ./squashfs-root/AppRun

Works!!!

probonopd, 2018-07-28:
So, FreeType should not be bundled @tibor-vavra