prusa3d/PrusaControl #115: libfontconfig.so.1: undefined symbol: FT_Done_MM_Var

mariansam, 2018-11-02:
I just downloaded the AppImage for Linux, when I try to run it, it says
```
Traceback (most recent call last):
  File "./PrusaControl-master/main.py", line 10, in <module>
  File "./PrusaControl-master/controller.py", line 21, in <module>
ImportError: /lib64/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var
```
for debugging:
```
marian@sheldon /tmp/prusactrl$ env LD_DEBUG=libs ./prusacontrol.AppImage 2>log
marian@sheldon /tmp/prusactrl$ grep font log
     24266:	find library=libfontconfig.so.1 [0]; searching
     24266:	  trying file=/tmp/.mount_prusaczfPkFC/usr/lib/libfontconfig.so.1
     24266:	  trying file=/tmp/.mount_prusaczfPkFC/usr/lib/x86_64-linux-gnu/libfontconfig.so.1
     24266:	  trying file=/tmp/.mount_prusaczfPkFC/lib/libfontconfig.so.1
     24266:	  trying file=/tmp/.mount_prusaczfPkFC/lib/x86_64-linux-gnu/libfontconfig.so.1
     24266:	  trying file=tls/x86_64/x86_64/libfontconfig.so.1
     24266:	  trying file=tls/x86_64/libfontconfig.so.1
     24266:	  trying file=tls/x86_64/libfontconfig.so.1
     24266:	  trying file=tls/libfontconfig.so.1
     24266:	  trying file=x86_64/x86_64/libfontconfig.so.1
     24266:	  trying file=x86_64/libfontconfig.so.1
     24266:	  trying file=x86_64/libfontconfig.so.1
     24266:	  trying file=libfontconfig.so.1
     24266:	  trying file=/lib64/libfontconfig.so.1
     24266:	/lib64/libfontconfig.so.1: error: symbol lookup error: undefined symbol: FT_Done_MM_Var (fatal)
ImportError: /lib64/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var
marian@sheldon /tmp/prusactrl$
```
However, I am able to run it with
```
LD_PRELOAD=/usr/lib64/libfreetype.so ./prusacontrol.AppImage
```
Info about my system:
```
marian@sheldon /tmp/prusactrl$ uname -a
Linux sheldon 4.18.16-300.fc29.x86_64 #1 SMP Sat Oct 20 23:24:08 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
marian@sheldon /tmp/prusactrl$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 29 (Twenty Nine)
Release:	29
Codename:	TwentyNine
marian@sheldon /tmp/prusactrl$ cd /usr
marian@sheldon /usr$ find . -name "*fontconfig.so*"
find: ‘./share/polkit-1/rules.d’: Permission denied
find: ‘./share/selinux/targeted/default/active’: Permission denied
find: ‘./libexec/initscripts/legacy-actions/auditd’: Permission denied
./lib64/libfontconfig.so.1.12.0
./lib64/libfontconfig.so
./lib64/libfontconfig.so.1
./lib/libfontconfig.so.1.12.0
./lib/libfontconfig.so.1
marian@sheldon /usr$
```

probonopd, 2018-11-02:
libfreetype should be removed from the AppImage. Try the following:

```
./PrusaControl_0.9.4_415_beta_64bit.AppImage --appimage-extract
find squashfs-root/ -name '*freetype*' -delete
./squashfs-root/AppRun 
```

Does it run now? If yes, then not bundling libfreetype in the AppImage will resolve this issue.

probonopd, 2018-11-02:
This is a duplicate of https://github.com/prusa3d/PrusaControl/issues/93.

mariansam, 2018-11-03:
Thank you so much! I will try to pack it back using https://github.com/prusa3d/PrusaControl/issues/93#issuecomment-431660150 - closing.

closed 2018-11-03 by mariansam